djgreeb/CDJ-1000mk3_new_life_project #1: STM32F746 Stuck on loading of logo

ArthurEschner, 2018-05-29:
Followed the install instructions, but the microcontrollers takes forever to load the logo, then it shows the version & progress bar rolls across the display and then it is stuck. It just doesn't do anything after that. Version 85 is the one im referring to. However V86 doesn't even complete loading the logo. Is there anything / settings that I have overlooked in the STUtily app? Any help will be really appreciated.

djgreeb, 2018-05-29:
> Followed the install instructions, but the microcontrollers takes forever to load the logo, then it shows the version & progress bar rolls across the display and then it is stuck. It just doesn't do [...]

The reason is found. The normal operation of the STM board is prevented by the signals ULPI_1 and ULPI_13. Cut the paths as shown in the picture in my manual (page 12).  I wrote the manual after cutting these paths, so I did not take into consideration that they would interfere with the loading in the new STM boards. Those who first cut the paths, and then downloaded the firmware, did not encounter this problem. Good luck!

ArthurEschner, 2018-05-29:
Totally sorted the issue out. Thank you so much!!

ArthurEschner, 2018-05-31:
Damn, I think I killed my Discovery board!!
It was working perfectly, but after I connected it to the cdj, the screen turned on green (still shows rekordbox though). It's almost like it's missing one layer. Occasionally the screen will display things normally again, but for the most part it's just green. Pc still detects it, but the display seems very dodgy :(  I have ordered another 2 discovery boards & will work on them with extra care next time. If you have any ideas what might be causing my screen issue, I'd appreciate your input. Thanks for your guidance. Regards.

djgreeb, 2018-06-01:
oh, I sympathize with your bad news. Download the demonstration firmware from the STmicroelectronics.com website and flash your STM32F746G board using the STLink utility, check if everything is working on your board (display, touchscreen, audio codec). If you do not find the demonstration firmware, write to me, I will send it to you.
What exactly did you already connect to the CDJ? What signals?
In your CDJ were some kind of defect before that?
Before connecting a new STM32F746G board, check that the CDJ outputs the required voltage (5V and 12V for vled). Did you install the 74HC245 chip? Make sure that this chip receives power from the board STM32F746G (3,3V) not 5V! it is important!

If in CDJ you have already deleted the original display, then it is necessary to apply a 5V 36ohm resistor.

Follow all the steps in the instructions sequentially, after each step, check if you have done everything correctly, only then go to the next step. All connections are made when your board and your CDJ are disconnected from the power.
Sorry, my English is bad, I use a translator.
Good luck to you!

djgreeb, 2018-06-01:
> Hi Andrei.
>
> Thank you for reply.
> Your english is actually very good!
> I followed your instructions every step of the way. I made sure I had
> stable 5v from cdj to power the ST32.
> I also took the 3.3v directly from the ST32 as per your diagram.
> I tried to load the demo firmware as you suggested and to my surprise...
> the Demo firmware seems to work fine.
> After loading the demo firmware, I can restart the ST32F7 over & over & it
> seems to operate normally.
> However when I load your cdj firmware, strange things happen....
> I've uploaded a video to help you to see & understand what my board is
> doing:
> https://vimeo.com/272919597
>
> Here is a breakdown of what is happening in the video:
> 1) I program the ST demo software, the ST32 restarts & the firmware
> displays normally
> 2) I then upload your V86 Pioneer firmware, the ST32 restarts but doesn't
> display Pionee Logo (strange green lines at the start), then the rekordbox
> screen shows normally.
> 3) I close the ST Utility software & restart the ST32, then the display no
> longer functions correctly.
> It only display correctly the 1st time after uploading the firmware.
> Anytime I restart it thereafter, the display looks damaged.
>
> I'm really not sure what's going on.
> Anyway, thank you so much for your time & help.
> Regards Athur.

Problems with the green display were always from the very first download or appeared later?
I'll think about this problem. If I have any ideas, I'll let you know.
And further. Write me what's written on your SDRAM chip, it's near the STM32F746G chip

ArthurEschner, 2018-06-03:
The problem with the green display appeared later, only after I connected it to the cdj.
It was working perfectly before that. However I don't think the touch screen has ever worked. When I bought the board I never actually tested the demo firmware. I just clicked 'erase' then loaded the Rekordbox firmware. When it was getting stuck on the logo at startup, I decided to load the demo firmware again, thats when I noticed the touchscreen wasn't working. This was before I cut any tracks or modified the board in any way. So I guess it's possible I bought a defective board. Unfortunately I cant return it now because I removed the camera connector & cut the 2 tracks. The SDRAM is 5JL12 (512mb)

djgreeb, 2018-06-04:
It is very strange. Memory, should be installed MT48LC4M32B2B5-6A (128mbit). On my board this is the capacity. In the evening I will check what is written on my chip. If Dicsovery installed 512mbit, it would be very good, I could add more functions (more hot cues, beat-jump), but on the official site there is 128-Mbit SDRAM (64 Mbits accessible) and the same number on my board.

If there is really another SDRAM on your board, this may be the cause of such strange work, you need to study its characteristics.

djgreeb, 2018-06-04:
On some boards there is another SDRAM. It is necessary to find out, maybe with your board everything is fine, just another revision.

ArthurEschner, 2018-06-04:
That is very interesting.
Perhaps I just read the data sheet incorrectly.
But when I googled the pdf for 5JL12 I was taken to this link: 512Mb: x4,
x8, x16 SDRAM - Micron Technology, Inc.
<https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwiXqcvk3rnbAhVrB8AKHZwjAi8QFggoMAA&url=https%3A%2F%2Fwww.micron.com%2F~%2Fmedia%2Fdocuments%2Fproducts%2Fdata-sheet%2Fdram%2F512mb_sdr.pdf&usg=AOvVaw0ACwi7XObelt80bwknCRxt>
I totally agree with you, that looking at the specs written on the
packaging it definitely says 128-Mbit

djgreeb, 2018-06-04:
Yes, I saw the same thing in google. I'll look for the marking rules, need to determine exactly which chip is soldered on the board.

djgreeb, 2018-06-04:
On my board is installed 5LL12. This is faster  166 MHz, but less capacious SDRAM (128mbit).  5JL12 (512mb) - is slower (100 MHz), and my project works at a frequency of 108 MHz. Perhaps because of this there are problems with the display. The demo firmware works at a frequency of 100 MHz, apparently in order to work on all board revisions. I will lower the frequency in my project to 100 MHz, although I would not like to do this.

ArthurEschner, 2018-06-04:
> cd=1&cad=rja&uact=8&ved=0ahUKEwiXqcvk3rnbAhVrB8AKHZwjA
> i8QFggoMAA&url=https%3A%2F%2Fwww.micron.com%2F~%2Fmedia%
> 2Fdocuments%2Fproducts%2Fdata-sheet%2Fdram%2F512mb_sdr.pdf&usg=
> AOvVaw0ACwi7XObelt80bwknCRxt >
> issues/1#issuecomment-394295173 >,
> AllRdhccoXKJft8XMbhGXCd59j0XtkbQks5t5QA6gaJpZM4UQ15q >

Wow, I wonder if there is any specific stock code to ensure that I get the
correct boards.
I have already ordered another 2, I think I will have a talk with the guys
at ST here in South Africa to see if they are aware that some boards differ.
Do you think the frequency will have a negative impact on performance?
If you can lower the frequency without too much effort as a test, that
would be great. However, it is a lot of work then rather leave it as it is.
I will rather ask the guys at ST to supply me with boards that have a
faster clock speed.
I'm sure a lot of guys are doing your upgrade, so I'm surprised no one else
has had this issue.

djgreeb, 2018-06-05:
I will lower the frequency and send you the firmware. I'll do this later, I need a few days for another job. The performance will drop by about 10%, but then this firmware will become universal, for all types of SDRAM.
PS. All this is in theory.  To check whether the SDRAM chip affects the operation of my project, it will be possible only after you download the test firmware.

ArthurEschner, 2018-06-06:
> issues/1#issuecomment-394429311 >,
> AllRdghKQWMulYWEYBnGITv0XrnEf45Jks5t5Wn_gaJpZM4UQ15q >

Hi Andrei.

Thank you, I really appreciate your help.
Whenever you get the time to do the firmware is fine. It's no problem at
all.
I will test & let you know the outcome.I'm also waiting for a response from
STmicro about the difference in boards.
Perhaps I purchased an old board, but as soon as I know for sure, I will
let you know.

Kind regards Arthur.

djgreeb, 2018-06-06:
>On the contrary, you have a new revision of the board The memory capacity is increased. It is also possible that your board is not original (China copy), since Stmicroelectronics shared all the documentation for the manufacture of these printed circuit boards in the factory.
>
>I send you a hex file of the firmware with a lower frequency. I checked, this firmware works on my CDJ, the performance decrease is not noticeable.
>Good luck!
>
https://drive.google.com/open?id=1kUWwbR60mzTnVU4PelT93ddwSI2POIBy

ArthurEschner, 2018-06-06:
Wow, that was quick!!.

I purchased the board directly from STmicro here in South Africa, so I
doubt it's a chinese clone.
However it was the only one they had in stock - their demo unit.
Unfortunately I think there's a problem with the DL Link.
I'm unable to download it - "Safari can't open the page"

djgreeb, 2018-06-06:
attached to the letter

ArthurEschner, 2018-06-06:
> issues/1#issuecomment-395188105 >,
> AllRdhjjjlZFNXjspGiVsbCEUhr9Bj-wks5t6C8rgaJpZM4UQ15q >

The attachment takes me to google drive,
when I try download = can't open page

djgreeb, 2018-06-06:
I understood this, and in the previous letter I attached the firmware file in the archive to the letter itself (not a link to google drive).
Attached again)

ArthurEschner, 2018-06-06:
> issues/1#issuecomment-395191381 >,
> AllRdias4MwyP6wHBrzWtd7IcrYtXQj0ks5t6DH4gaJpZM4UQ15q >

Haha, Im sorry about that.
Each time I clicked download - it took me to google drive.
I had to right click & download file as...
Now I feel stupid - Lol!!

djgreeb, 2018-06-06:
so it's okay))

angusmack, 2019-01-20:
@djgreeb - do you think it is always necessary to cut the tracks ULPI_1 and ULPI_13 to make the firmware load correctly on the newer boards? Would it be possible to work around this in the code? I'm worried about damaging my board (and also stopping me using this board for a different project in the future). Thanks