djgreeb/CDJ-1000mk3_new_life_project #2: Source Code

DJDan, 2018-07-03:
I can't seem to find the CDJ1000 mod source code anywhere. There is just the compiled firmware (.hex) file inside the zip.

Could you please put your latest source code on GitHub directly so that others in the community (like myself) can modify/improve the product?

There is options that are things coming in future firmware (like USB HID MIDI and USB/.MP3 playback) that people need that could come quicker if others could add to source code too.

Thanks

fer22f, 2018-07-13:
Please consider this, it could start an amazing community effort towards CDJ modding.

The work you have done here is beyond impressing; and you can protect your authorship by using an appropriate license. You can limit commercial use of your software, or make it forcely free using [GPLv3](https://choosealicense.com/licenses/gpl-3.0/) for example, which means everyone using a derivative of your firmware has to disclose the source code and reference your work.

beerjongen, 2018-09-22:
Please make it public, i'm working on a similar project, and I would love to learn from your findings!!

angusmack, 2018-09-22:
I would also like this. I've bought the electronic parts for 2 units and I'm looking forward to getting this working. 

I am planning to model the case in 3D and share the STL files once I've done it as a small way of giving back.

I imagine USB support along with other audio formats will be most requested. I hope we will all be able to contribute to this great project.

ecutsa, 2019-01-06:
andrei released the source code

angusmack, 2019-01-07:
Thanks for sharing. Much appreciated.

djgreeb, 2019-01-11:
Good luck!

>
>Thanks for sharing. Much appreciated.

Lukasguy, 2020-03-31:
Hi
Would it be possible to update the source code to the new version that was recently released? 
I'm currently adding hot cues and loops, and would like to have this on the new version as well. 

Many thanks for starting this project, making a good manual and releasing source code! I'm having a great time with this project!

floxia, 2020-04-15:
Anybody has successfully added the loop feature and wants to share the firmware?

closed 2020-07-26 by DJDan

vazhnov, 2020-08-08:
@fer22f , please see https://github.com/djgreeb/CDJ-1000mk3_new_life_project/issues/13 for the project license.