djgreeb/CDJ-1000mk3_new_life_project #4: Problem compiling sorce code

MarceloCampos, 2019-01-15:
Dear Andrei,

First, you did a great job in this project, congratulations.

I´m having problem to compile the code, and think this is because Reset_Handler (reset vector) is missing, attached image of compile result

[image]

thanks !

angusmack, 2019-01-20:
I'm not sure about this error. I was able to compile and program the source code as it was provided. I loaded the F7.uvprojx project file. Looks like you are doing something different with your project files.

[image]

Just changed the firmware version as a test of programming:

[image]

MarceloCampos, 2019-01-25:
thank u angusmack, now is Ok but, the code traps on "demo limit" code size ...

angusmack, 2019-01-27:
I have forked the repo and uploaded the source code https://github.com/angusmack/CDJ-1000mk3_new_life_project

I am in the process of building the electronics hardware. Once I have that I will be able to develop the code. I have had a good look at it and it looks ok to me.

Please, if you want help then raise issues on my fork and I will be able to help better. 

Thanks,
Angus

angusmack, 2019-01-27:
I need the hardware so I can make the STM32 work, otherwise it does nothing!!

AIFF and MP3 support are first on my list. Please feel free to request more features on my forked repo.

vincentdallamacchia, 2019-01-28:
hello I would like to know if the loops and hot cues will work with this mod on cdj 1000mk3 and then it would be nice if there was a loop roll section via touchscreen

angusmack, 2019-01-28:
I will work on new features in the software once I have modded my cdj1000mk3 player with the electronics.

This will not happen next week!! Please be patient :)

I have uploaded the source to my repo so other people can work on it if they have the hardware already.

I want to keep this project alive also.

jrsmi9, 2019-01-29:
Hi all, I have basically finished the modification on one cdj, aside from the cdj leds not working (trying to fix that at the moment), and no vincentdallamacchia, currently there are no loops or hot cues in the software from what i can tell. 
Unfortunately I have no idea how to write code :( 
angusmack, I know it will take a while, but a great feature to include once you are finished would be the ability to sort by playlists from rekordbox, as well as be able to read mp3 tags once that is supported (currently it displays "House" for every song genre).
For the case and buttons, i had them 3D printed and i can email anyone the stl files if you want them

[image]
[image]
[image]

vincentdallamacchia, 2019-01-29:
ok angusmack looking impatiently :)) this project has impressed me from the first moment but unfortunately I'm just an electronic expert and not a programmer so I'm waiting for your news ... I've designed some improvements on the type of hardware for the case I'm going to use this coverage original of the cdj 900 the code is dnk5442 fits very well on the display by 4.3 of the st ... and then I'm also using the original keys of the cdj 900 code dac2480 for the keys tagtrack and back then code dac2482 for the device keys and finally dac2492 for the browser keys utility taglist and info ..... comes a very clean job and for the sd card I'm using the slot of the front of the sd card of the same cdj ...... these are all hardware improvements .... .for the software improvements wait for you angus :))) greetings soon I hope :)

vincentdallamacchia, 2019-01-29:
would be nice a performance section on the touchscreen display that would implement auto loops, loop roll and hotcue .... otherwise the software is fine even with wav files that in my opinion are the best sound quality and are analyzed even better by rekordbox having bitrate fixed .... the only flaw is that if you have many tracks of course is a problem of space ...

angusmack, 2019-01-29:
I think USB support would be good too. Connect micro USB cable from STM to female USB standard socket exposed on case so you can use a USB stick as well as SD card.

I use AIFF so support for that format as well as WAV (very similar) would be nice :)

I just need to make the PCB for the buttons and encoder wheel so I can interact with the STM first...

vincentdallamacchia, 2019-01-29:
ok wait for updates :)))

angusmack, 2019-01-29:
@jrsmi9 - please can you push the STL files for the case and buttons to my repo https://github.com/angusmack/CDJ-1000mk3_new_life_project in a new folder called "STL Files". If you don't know how to do this then you can email them to me angus dot mack at gmail dot com and I will do it. Thanks.

vincentdallamacchia, 2019-01-29:
however I doubt that you can read mp3 with this card stm ... as a format read only the wav
it would take an mp3 decoder

vincentdallamacchia, 2019-01-29:
angus a question using a mini usb to usb female adapter can you already read the usb sticks or still not able ??? then I saw that the buttons work even without the resistors and the capacitors only by connecting them to the ground and instead of the stm of competence except the encoder that needs the resistors of the signal antidebunge

jrsmi9, 2019-01-30:
I've emailed them, check spam folder too perhaps

angusmack, 2019-02-06:
Thanks. I got them and I have added them to my repo.

vincentdallamacchia, 2019-02-09:
angus i wanted to know is there an easier way to pick up the 4 ulp signals 1 and 13 and camera 15 and 17 from the stm board ??? the tracks are very thin and sometimes you can not weld on them even if you have a magnifying glass :(

jrsmi9, 2019-02-09:
For the ulp 1 and 13  I cant help, I managed to solder them correctly
For the Camera 15 and 17 you can use a ribbon cable breakout board such as this one below which makes it a lot easier:

Breakout Board
https://www.ebay.com/itm/1pcs-30-Pin-0-5mm-FFC-FPC-to-30P-DIP-2-54mm-PCB-Converter-Board-Adapter/252986447855?hash=item3ae72ad7ef:g:7j8AAOSwnK9ZPeHA:rk:2:pf:0

Ribbon Cable
https://www.ebay.com/itm/FFC-FPC-Flexible-Flat-Cable-Ribbon-30-Pin-Forward-Reverse-Pitch-0-5mm-1-0mm-/263425258855

jrsmi9, 2019-02-09:
However I am unsure if I have done the camera connections correctly as the LED on my CDJ still do not light up. Angus I am wondering if you can tell from the source code which conections are responsible for signalling the CDJ LED's to turn on, so that I can determine where I have gone wrong?

vincentdallamacchia, 2019-02-10:
thanks I will try to weld the wires ulp 1 and ulp 13 hoping not to do damage and another question there are news about hot cues and loop roll ?? as soon as I finish I will try to post the photos with the original pieces looks like a cdj 900 nexus .... ah another question is possible with this card use a 7-inch screen like that of raspberry also without touchscreen .... I saw of an Italian guy who used raspberry by installing the program directly on the board and can be used with any cdj .... raspberry is much more complete as inputs and many boards of arduino are compatible with it and costs much less than the stm ..... however, rekordbox and its layout are always my favorites:);) ..... I am impatient for the hot cue firmware .... in these days I noticed that loading a track automatically opens the tag list menu and makes a kind of flashing until I hit the browser button or else .... it happened to some of you ??? sorry if I have harassed ahahahah

vincentdallamacchia, 2019-02-10:
sorry this poor english but I use google translate .....

djgreeb, 2019-02-25:
Unfortunately I did not work with Rasoberry, perhaps it would be a better solution than STM. There were no problems with the TAG button. Check the solder wire and button.

> thanks I will try to weld the wires ulp 1 and ulp 13 hoping not to do damage and another question there are news about hot cues and loop roll ?? as soon as I finish I will try to post the photos [...]

djgreeb, 2019-02-25:
It is not possible to output all signals through a ribbon cable. ulp 1 signal with hardware interrupt, it can not be reassigned to the camera connector

> For the ulp 1 and 13  I cant help, I managed to solder them correctly
> For the Camera 15 and 17 you can use a ribbon cable breakout board such as this one below which makes it a lot easier:
> Breakout [...]

djgreeb, 2019-02-25:
I was not involved in the implementation of software mp3 decoder. This will be the last step. It all depends on how much free resources remain in the processor.

>
>however I doubt that you can read mp3 with this card stm ... as a format read only the wav
>it would take an mp3 decoder

djgreeb, 2019-02-25:
The front port on the SD card (CDJ-1000mk3) is too far from the STM, I'm not sure if it will work well at high speeds, there may be problems with interference.

> ok angusmack looking impatiently :)) this project has impressed me from the first moment but unfortunately I'm just an electronic expert and not a programmer so I'm waiting for your news ... I've [...]

djgreeb, 2019-02-25:
Congratulations! good job! I posted the source code in public, I hope this will speed up the process of improving the firmware by other users.

> Hi all, I have basically finished the modification on one cdj, aside from the cdj leds not working (trying to fix that at the moment), and no vincentdallamacchia, currently there are no loops or hot [...]

djgreeb, 2019-02-25:
Do you have problems with backlighting buttons on the CDJ or with panel buttons (BROWSE, TAG LIST, INFO ...)? If the problem with the buttons on the CDJ, jog, pitch work? If yes, then check CAMERA_15 -> SIN

> Hi all, I have basically finished the modification on one cdj, aside from the cdj leds not working (trying to fix that at the moment), and no vincentdallamacchia, currently there are no loops or hot [...]

djgreeb, 2019-02-25:
Hi, this is Andrew. I do not have much time to engage in this project, and not a large number of people have an interest in it (I think so). 0.86 is the latest update for the moment, but I posted the source code of the firmware publicly, I hope this will speed up the development

>
>hello I introduce myself vincenzo ... I would have questions about this project ... I want to turn my 1000mk3 but I would like to know if with the latest firmware 0.86 the loop session works and also the session hot cue ... I tried to contact andrei but no longer responds to any comment .... it would seem that he abandoned the project .... there is the hope of other firmware with additional features such as a loop roll via touchscreen and a quantize and that the session really work ?? It's a nice project and it would be a shame to leave it half way ... thanks to everyone

vincentdallamacchia, 2019-02-26:
pity Andrei because I think you're the only one who could take a project of this .... I took a look at your other projects and I must tell you that if I had your skills I would not leave aside so .... I'm an electronic expert but unfortunately I can not make do with the programming .... so I ask your help and that of angus to carry out this project that excites me .... it would be nice if it worked the loops and hot cues. ... even if you read the wav without mp3 .... great work Andrei

angusmack, 2019-03-01:
Originally I forked the repo so I could work on it but this meant search was not working properly so I have made a copy here https://github.com/angusmack/CDJ-1000Mk3-MCU-Mod

If you require any support or assistance or want to ask questions please do so in this repository so we can all share what we have.

To be clear Vincent: although I am very capable to create the features you are asking for, the time I can dedicate to this project is small and I will do what I can when I can. If anyone else wants to submit pull requests to the repo then please feel free.

I am looking forward to making this work so please Star/Follow the repo and share.