djgreeb/CDJ-1000mk3_new_life_project #7: Digital audio noise in end track

fabiodj2, 2019-03-21:
Hello everyone ... I've done my design step and it's almost all working ...

I am printing the case for the STM board and very eager to get everything ready. I was testing all the functions and I noticed a big one, I believe that it could be some problem in my STM, so I would like you to do a test in the same way that I like in the video below:

https://youtu.be/iJTq9NwelFo

The problem is a digital noise whenever the Tracker is finished and the CDJ Pitch is out of 00.00%.

Has anyone else had this problem?

Thank you Very much!!!

djgreeb, 2019-03-31:
Hey. I understand what it is about. This noise is heard because the last audio frame continues to play. This is not an error, I just have not added the firmware yet and have not closed this bug.

> Hello everyone ... I've done my design step and it's almost all working ...
> I am printing the case for the STM board and very eager to get everything ready. I was testing all the functions and I [...]

fabiodj2, 2019-03-31:
Hi...thank you very much for your feedback...do you have any forecast when we will have a new firmware with new functions? I've made some changes to the case design and I'm going to do a few more things in the electronics. As soon as I finish I can share the changes and ideas with you! Of course if it is your wish! On the case of colored Waveformes, would colored candles be possible?

djgreeb, 2019-04-14:
Hi. I can not name the timing. Last time I was not involved in this project.

> Hi...thank you very much for your feedback...do you have any forecast when we will have a new firmware with new functions? I've made some changes to the case design and I'm going to do a few more [...]