djgreeb/CDJ-1000mk3_new_life_project #12: cdj 1000 mk3 firmware mod to play wav/aiff/mp3

janhase88, 2020-04-23:
hi,
i just found this project on youtube - really great project! 
I wonder, is it possible to just mod the firmware (no hardware changes/physical modding) so the mk3 can play wav/aiff & mp3 from sdcard ?

*sry this is probably the wrong place to ask, but i couldnt find your email :-)

cheers

sashabeep, 2020-05-02:
@janhase88 IMO, board power is too weak to handle decoding and all of the interface things at the same time with stable result.

@djgreeb correct me if i wrong. Have you been tried helix decoder?

djgreeb, 2020-05-14:
Hi, maybe the resources of the original processor in the CDJ-1000 are enough to play wav and mp3, but to change the firmware, you need to get the source codes for this firmware, and this is pioneer's personal information.

> 
> 
>hi,
>i just found this project on youtube - really great project!
>I wonder, is it possible to just mod the firmware (no hardware changes/physical modding) so the mk3 can play wav/aiff & mp3 from sdcard ?
>*sry this is probably the wrong place to ask, but i couldnt find your email :-)
>cheers

closed 2020-07-10 by janhase88