djgreeb/CDJ-1000mk3_new_life_project #17: Error 15

wussuphaters, 2021-01-22:
Hi ! Thanks a lot first of all, I gave this project a try and managed to do it. I even designed a custom PCB for the buttons etc, making it more convenient. Only problem is, when I load a track, it says error 15 and waveform does not appear. I can still see that SPI is indeed working because i can change tempo etc, but cannot play the track. I used .mp3 converted into .wav, I suspect it is a format error, can you detail this ?

antonio-mx, 2021-01-31:
You're right, the same thing happened to me and it was because of the audio format, it only supports 44.1k 16bit stereo

DJJuciel, 2021-02-02:
hello wussuphaters, how are you? I hope everything is alright!
sorry my english, because i'm using google translator!
I'm starting my project, but on the electronics I'm still an amateur!
Did you mount the SPI LEVEL SHIFTER card? and the custom PCB for the buttons you mentioned?
Would you be able to send me, or the file or photos of how you made these PCBs, will be of great help to me! if you can send me, this is my email: deejayjuciel@gmail.com
Thanks in advance!

wussuphaters, 2021-02-03:
Indeed it was due to a format error ! It is a pretty quick fix using Audacity to convert a bunch of .mp3 files to .wav and then resample them before exporting.

@DJJuciel below you will find a link to the PCB files, you can use JLCPCB or PCBWay to make them, I used JLCPCB and it cost me about 12€ for 5 of them. The PCB is basically a hub for all the components, there are 4 pin headers that are labeled DISCO (connection to Discovery board), CDJ (SPI connexion to CDJ), BUTTONS (button outputs to Discovery board) and LEDS (inputs for LED switching). Individual pinout are not labeled but you can use the .sch files (Eagle or Kicad software) to identify them.

Also make sure to order the right component footprints, for example the bus transceiver IC is SOP20 and not TSSOP20, which is smaller.

This PCB is far from being perfect but I hope it can be of use to some who would like to recreate the project ! It has been working fine for me :)

https://drive.google.com/drive/folders/12zg70nERy2KB2MboOX39imnLJUJ70jv0?usp=sharing

closed 2021-02-03 by wussuphaters