djgreeb/CDJ-1000mk3_new_life_project #18: SPI level shifter

Stevieheff, 2021-02-17:
Hey guys :) So i have decided to attempt this project. After piecing together all my parts i am struggling with soldering the 74HC254 chip to the breakout board. I am using 0.6mm solder with a fine tip. are there bigger components that i can use instead of the smd chip? I am excited to attempt this project and i would appreciate if you guys had any photos from during your build that i can reference? :)

Letrab, 2022-04-01:
Here is an explanation of a (through hole) SPI level shifter as well, probably compatible? https://www.vandoeselaar.com/tinkering/cdj-1000-mk3-for-rekordbox/

mtoonk, 2024-09-09:
@Stevieheff did you managed to make the level shifter? As a n00b I also find it very complex to make it. Does someone have clear pictures of the boards in relation to the schemes. Maybe someone used a breadboard?