djgreeb/CDJ-1000mk3_new_life_project #20: test music playback without browse buttons

mdethmers, 2021-07-10:
hey, first of all, this project is amazing! love the idea of upcycling such a product!

I have bought everything needed for the mod, and have it all connected now. the buttons all seem to work (bpm, quantize, time etc.) but I have no idea how to load a track on it! 

i just have the screen connected, and loaded with a 44.1khz 16bit wav file, but pressing the search track or search button does not appear to be doing anything! also no errors on the screen etc... the self assembly guide says that you should be able to use the track search to load tracks, butit does not seem to work? I don't think its a connection error as the rest works fine. any thoughts on this?

thanks!

fabiodj2, 2021-07-12:
Hello friend....I'm glad you also decided to do this project...could you
send me a video of what's going on? It would be easier to tell you

mdethmers, 2021-07-12:
hey fabio, thanks for coming back to me and willing to help me out! 

I've made a video for you which shows what works/does not work. 

https://www.youtube.com/watch?v=oDTlho7wmc0

the sd is loaded with a 44.1khz 16bit sample song as well! is there maybe another version from which I could try to load the song?

fabiodj2, 2021-07-13:
OK my friend...now I understand....You will need the encoder to load and
browse the songs and also the project browser buttons.

[image: image.png]

https://www.ebay.com/itm/Brand-New-Rotary-Encoder-EC11-Digital-Potentiometer-Handle-15mm-20mm-Kit-Set/222355519139?hash=item33c56bf6a3:m:mMKi-XWIttuvhMb6FlXcfCg

https://www.ebay.com/itm/100pcs-3X6X2-5mm-Tactile-Push-Button-Switch-Tact-Switch-Micro-Switch-2Pin-SMD-S2/261995936109?epid=1046591979&hash=item3d002cb96d:g:RSsAAOSwyQtVxKPd

The CDJ project uses in a mixed way the original keys from the panel of the
CDJ-1000 MK3 and also the keys "created" by the author of the project.

If you have difficulty with the electronics part and want a plug and play
item from the project...I've finished here the cases, buttons, keys and
other items.

[image: image.png]

Instagram...

https://www.instagram.com/explore/tags/cdj1000nexus/

https://www.instagram.com/p/CM-pbR3FhfP/

Best Regards,

Fabio Lisboa

adi2870, 2025-01-04:
> hei fabio, mulțumesc că ai revenit la mine și că ești dispus să mă ajuți!
> 
> Ți-am făcut un videoclip care arată ce funcționează/nu funcționează.
> 
> https://www.youtube.com/watch?v=oDTlho7wmc0
> 
> sd-ul este încărcat și cu un eșantion de cântec de 44,1 kHz pe 16 biți! poate exista o alta versiune din care as putea incerca sa incarc melodia?

> hei fabio, mulțumesc că ai revenit la mine și că ești dispus să mă ajuți!
> 
> Ți-am făcut un videoclip care arată ce funcționează/nu funcționează.
> 
> https://www.youtube.com/watch?v=oDTlho7wmc0
> 
> sd-ul este încărcat și cu o eșanție de cântec de 44,1 kHz pe 16 biți! poate exista o alta versiune din care as putea incerca sa incarc melodia?

Hello, mdethmers, I also started this project, very interesting, but my cdj1000mk3 does not start.
Start the board with stm32f, I can select the song from the buttons, but the player does not start.
Do you have any idea, I see that your CDJ is starting.
Thank you!

mdethmers, 2025-01-07:
Hey! I am not sure, but in the end all th functions of the player only worked after I added/connected all the peripheral buttons and sliders. make sure everything is connected and functioning, as the play-button might not be connected now and therefore does not work.